jczic/MicroWebSrv #3: Post example on ESP32 not working

rdagger, 2017-09-04:
The example _httpHandlerTestPost method fires but doesn't make it past the line formData = httpClient.ReadRequestPostedFormData().  There is no error message.  Looks like the ReadRequestContent method is the culprit.  The response stops in this method at b = self._socket.readall().

jczic, 2017-09-04:
Thank you for your feedback.
The readall() method waiting a closing connection.
I updated by reading the ContentLength size now.
Could you try on your ESP32 with the same browser and tell me the result please ?

rdagger, 2017-09-04:
It's working now.  Great Job!

2 small notes in microWebSrv.py:  
_sys.exc_info()_ is not implemented (at least not on ESP32, not sure about ESP8266).
_uhashlib_ is not implemented either and it does not appear to be used in the code.

Also could you please post an example of how to serve existing files (html, png, pdf, etc.)

closed 2017-09-04 by rdagger

jczic, 2017-09-04:
Thank you :)
Yes, sys.exc_info () is on pycom modules (www.pycom.io) to show stack exceptions.
Exactly for uhashlib: I forgot to remove it after integrate websockets handshake (future extension) ...

To directly serve existing files, this is already okay with main.py: copy the "www" directory in your flash and go to your ESP by http://ip-addr/ to get the index.html with image, favicon and attached pdf file download.