jczic/MicroWebSrv #4: can't convert 'bytearray' object to str implicitly

rdagger, 2017-09-04:
I am having trouble running the example with regard to browsing index.html on an ESP32.
At first the index.html file was not found so I specified webPath='/www/' 
This fixed the file not found error.

But then I received a NameException:  name _'_tryAllocByteArray' is not defined_ in the _ResponseWriteFile()_ method.  I fixed the error by moving the function into the Response class.

Next I received the Exception: _can't convert 'bytearray' object to str implicitly_ in the same _ResponseWriteFile()_ method when it called the _write()_ method.  Looks like _client._socket.send(data)_ expects a string and not a byte array.  Normally I would convert with decode('utf-8') but it does not appear to be implemented.  Therefore I did an ugly hack of the _write()_ method:
```
if type(data) == str:
                return self._client._socket.send(data)
            else:
                for b in data:
                    self._client._socket.send(chr(b))
```
The page is now loading as HTML with CSS formatting.  However, the image is not loading.  My _write()_ method hack is throwing an [Errno 104] ECONNRESET.   I'm don't know why.

jczic, 2017-09-05:
Ok for "/www/" directly. Just, on pycom modules, all files are in
'/flash/...'.

I made an error today in the call of *tryAllocByteArray from _response
class *without test... I've fixed that.
I've moved all utils function to a "static" logic in the class. Same for
the public HTMLEscape() func.
Now, call it like MicroWebSrv.HTMLEscape(...) directly.

The socket.send problem to using buffer protocol is not present on pycom
modules, but I've changed that to socket.write.
Could you try that on your esp32 (data can be str, bytes, bytearray,
memoryview, ...) and tell me if that's works ?

rdagger, 2017-09-05:
Works great!  Runs fast too.
I totally appreciate your help.
Thanks.

closed 2017-09-05 by rdagger

jczic, 2017-09-05:
Yeah :)
I totally appreciate yours feedback too !