jczic/MicroWebSrv #5: MemoryError

RonaldKrause, 2017-09-05:
Hello, just tried MicroWebSrv on w WiPy 2.0 and got this error (with the unaltered files):

Traceback (most recent call last):
  File "main.py", line 1, in <module>
  File "microWebSrv.py", line 21, in <module>
  File "microWebSrv.py", line 98, in MicroWebSrv
MemoryError: memory allocation failed, allocating 1784 bytes
MicroPython v1.8.6-760-g90b72952 on 2017-09-01; WiPy with ESP32

Its the line: "_hextochr = dict(('%02x' % i, chr(i)) for i in range(256))"

I could get it running when I uncommented "_hextochr ", but only the URL "test" worked. index.html gave me a 404. 

Any idea whats wrong? 
(On peut communiquer en francais si vous voulez)
Merci, Ronald Krause

jczic, 2017-09-05:
Ok pour le française :)
Il va falloir qu'on comprenne cela ensemble alors.
Ce n'est pas normal que cela pose un souci de mémoire à moins que vous ayez
tout un tas d'autres choses chargées en mémoire.
Le WiPy n'a que 512 Ko qui sont en grande partie consommée de base...
A quel moment exactement l'erreur de mémoire intervient ?
Lors de l'importation de MicroWebSrv ou bien durant son utilisation ?

Pouvez-vous afficher cela avant que l'erreur se produise :

import gc
print(gc.mem_free())

RonaldKrause, 2017-09-05:
Alors...

C'est une installation "out of the box", du fichier zip sans changements. 
L'erreur s'passe en lingne 1 dans main.py, c'est "import  MicroWebSrv".

J'ai rajouté gs.mem_free  plusieurs fois:
1. boot.py début 
2. boot.py fin
3. main.py début 
4. microWebSrv.py ligne 97, avant 
    "_hextochr = dict(('%02x' % i, chr(i)) for i in range(256))"
   --> crash

Le Output: 
```
MemFree Boot1:  59504
MemFree Boot2:  59200
MemFree Main1:  56592
MemFree WebSrv1:  26080
Traceback (most recent call last):
  File "main.py", line 4, in <module>
  File "microWebSrv.py", line 21, in <module>
  File "microWebSrv.py", line 98, in MicroWebSrv
MemoryError: memory allocation failed, allocating 2344 bytes
```

J'ai aussi testé sur la REPL:
```
>>> import gc
>>> print(gc.mem_free())
59296
>>> _hextochr = dict(('%02x' % i, chr(i)) for i in range(1))
>>> print(gc.mem_free())
58672
>>> _hextochr = dict(('%02x' % i, chr(i)) for i in range(256))
>>> print(gc.mem_free())
```
36960

Ètonnant, n'est-ce pas?

J''ai alors remplacé le dict par une fonction: 
```
    def _hextochr(hexString):
        return int(hexString, 16)
```
Maintenant c'a l'air de marcher. 
Mais je ne pas reussi de voir "index.html". Comment faire?

jczic, 2017-09-05:
Ok, this is a bug of pycom module....
Problem to allocate memory for dict(('%02x' % i, chr(i)) for i in
range(256)).
Your function is wrong because it must be a dict with an str hexadecimal
index.
I'm checking for memory and this bug...

jczic, 2017-09-06:
Could you retry, I updated the method to bypass the memory problem.

2017-09-06 0:02 GMT+02:00 Jean-Christophe Bos <jczic.bos@gmail.com>:

> Ok, this is a bug of pycom module....
> Problem to allocate memory for dict(('%02x' % i, chr(i)) for i in
> range(256)).
> Your function is wrong because it must be a dict with an str hexadecimal
> index.
> I'm checking for memory and this bug...
>
>
>> Alors...
>>
>> C'est une installation "out of the box", du fichier zip sans changements.
>> L'erreur s'passe en lingne 1 dans main.py, c'est "import MicroWebSrv".
>>
>> J'ai rajouté gs.mem_free plusieurs fois:
>>
>>    1. boot.py début
>>    2. boot.py fin
>>    3. main.py début
>>    4. microWebSrv.py ligne 97, avant
>>    "_hextochr = dict(('%02x' % i, chr(i)) for i in range(256))"
>>    --> crash
>>
>> Le Output:
>>
>> MemFree Boot1:  59504
>> MemFree Boot2:  59200
>> MemFree Main1:  56592
>> MemFree WebSrv1:  26080
>> Traceback (most recent call last):
>>   File "main.py", line 4, in <module>
>>   File "microWebSrv.py", line 21, in <module>
>>   File "microWebSrv.py", line 98, in MicroWebSrv
>> MemoryError: memory allocation failed, allocating 2344 bytes
>>
>> J'ai aussi testé sur la REPL:
>>
>> >>> import gc
>> >>> print(gc.mem_free())
>> 59296
>> >>> _hextochr = dict(('%02x' % i, chr(i)) for i in range(1))
>> >>> print(gc.mem_free())
>> 58672
>> >>> _hextochr = dict(('%02x' % i, chr(i)) for i in range(256))
>> >>> print(gc.mem_free())
>>
>> 36960
>>
>> Ètonnant, n'est-ce pas?
>>
>> J''ai alors remplacé le dict par une fonction:
>>
>>     def _hextochr(hexString):
>>         return int(hexString, 16)
>>
>> Maintenant c'a l'air de marcher.
>> Mais je ne pas reussi de voir "index.html". Comment faire?
>>
>>
>
>
>
>
> Sincèrement,
>
> *Jean-Christophe Bos*
> *jczic.bos@gmail.com <jczic.bos@gmail.com>*

RonaldKrause, 2017-09-06:
It works, thank you!
Still memory is quite low.
I've printed mem at the end of  "_httpHandlerTestPost" here:
```
Start main:  55040
after import MicroWebSrv:  23200
mem:  12176
mem:  1616
mem:  17312
mem:  6064
mem:  13696
mem:  4160
mem:  19472
```
And one more question if you permit: how can i access the page "index.html"?

jczic, 2017-09-06:
Yes, the memory management on the pycom with GC
(collect/mem_free/mem_alloc) appears strangely sometimes ...
I had funny cases, hence the functions _tryAllocByteArray and
_tryStartThread.

Normally to get the index, you just need to go to http://ip-module/ and it
should go look for it automatically in /flash/www/index.html.
What is the problem about that or what does your browser display ?

RonaldKrause, 2017-09-06:
Its a "404 Not Found
Nothing matches the given URI".

The files are present on flash.

jczic, 2017-09-07:
Hmm strange...
Could you try (if IP is 192.168.4.1) :
http://192.168.4.1/test
http://192.168.4.1/favicon.ico
http://192.168.4.1/style.css
To check if this is not a memory problem again to read/serve index.html.

RonaldKrause, 2017-09-08:
Ok, I just now found out that I have to configure Atom/pymakr to synchronize all types like html, css and so on. I got it working now, it shows index.html. 

But I have two more quesions / problems: 
1. I did not manage to have all files in the www-subdir synchronized. It typically fails after the first file from the www-subdir:

```
Syncing project (main folder)...
Reading file status
[1/6] Writing file www/favicon.ico
Synchronizing failed: undefined. Please reboot your device manually.
```
I tried on two WiPy'S with different firmware, from Windows and Linux. 
Refreshing the  filesystem with os.makefs('/flash') does not solve the problem. 
At the same time other projects with subdirs work (with just *.py-files). 
Did you ever have any similar problems?

2. When main.py runs I cannot CTRL-C to stop it. So far I had every time to "safe boot" with Pin12. Is this normal, and why? 

Best regards and thanks for your time and the wonderful project

jczic, 2017-09-08:
1.
I never use Atom/pymakr.
Why do you not use FTP directly to upload all your files over it ?

2.
Yes, it's normal, in the main.py example, TCP(http) server is not threaded.
Remove threaded arg in main.py (server is threaded by default) :
srv.Start(threaded=False)   -->   srv.Start()

RonaldKrause, 2017-09-08:
Got it running now, thank you.

jczic, 2017-09-09:
:)
Merci pour avoir aidé !

closed 2017-09-09 by jczic