jczic/MicroWebSrv #7: exc_info not implemented on ESP32

rdagger, 2017-09-11:
sys.exc_info is not implemented for the ESP32 yet.  It is used in several modules in this repo.
Would it be OK to replace them with Exception.message such as:

```
except :
    print("MicroWebSocket : Error on accept callback (%s)." % exc_info()[1])
```
replace with
```
except Exception as e:
    msg = e.message if hasattr(e, 'message') else str(e)
    print("MicroWebSocket : Error on accept callback (%s)." % msg)
```

jczic, 2017-09-11:
Yes, I'll fix that :)

jczic, 2017-09-11:
I've removed the usage of exc_info :)

closed 2017-09-11 by jczic

rdagger, 2017-09-11:
Thanks!  
I'm making a series of ESP32 MicroPython tutorials on YouTube.  
Is it OK if I include your MicroWebSrv library in one of my videos?

jczic, 2017-09-11:
Thanks to you !
Yes, no problem, it's very cool !

Do you have already some videos to see ? :)

rdagger, 2017-09-11:
So far, I've made 2 videos.  Here's a [link](https://www.youtube.com/playlist?list=PLKGiH5V9SS1hUz5Jh_35oTFM4wPZYA4sT).

jczic, 2017-09-11:
Good job 👍

jczic, 2017-10-10:
Hello, do you have made your video with MicroWebSrc ? :)
2 news libs for querying HTTP(S) :

https://github.com/jczic/MicroWebCli
AND
https://github.com/jczic/MicroRESTCli

If you want to try on your ESP32 :)

rdagger, 2017-10-11:
I did record the video.  I just need to finish the video editing.  I'll probably post this week.
Those libraries look great!

rdagger, 2017-10-15:
I posted the [video tutorial](https://youtu.be/xscBwC1SrF4).

jczic, 2017-10-16:
Hi,

Yeah, I've seen, it's very cool !! :)

jczic, 2017-10-17:
How many RAM have you in your esp32 ?
My last OEM pycom modules have 4 Mo : excellent for SSL, multi-threading
etc...!

2017-10-16 21:46 GMT+02:00 Jean-Christophe Bos <jczic.bos@gmail.com>:

> Hi,
>
> Yeah, I've seen, it's very cool !! :)
>
>
>
>
>> I posted the video tutorial <https://youtu.be/xscBwC1SrF4>.
>>
>>
>
>
>
>
> Sincèrement,
>
> *Jean-Christophe Bos*
> *jczic.bos@gmail.com <jczic.bos@gmail.com>*

rdagger, 2017-10-18:
4MB flash, memory-mapped to the CPU code space.  I think there is an additional 520K of integrated SRAM.
Here is a link to the board:. https://wiki.wemos.cc/products:lolin32:lolin32

jczic, 2018-08-20:
Hello rdagger :)

After WebSockets in MicroWebSrv (https://microwebsrv.hc2.fr) you can see an
another microPython/ESP32 lib :
https://github.com/jczic/MicroMLP
(https://micromlp.hc2.fr)
This is more difficult ;)

Have a good day !

rdagger, 2018-08-20:
@jczic  Very impressive work!  Thanks.

jczic, 2019-10-20:
Hello rdagger :)

Do you remember microWebSrv and your great video on it?
I released a fully new version (v2.0) of my web server here :
github.com/jczic/MicroWebSrv2.
Open source MIT, fully asynchronous, more robust, more fast and more
efficient!
It is delivered with a good documentation.

Thank you for your support and feedback. ☺️

Jean-Christophe Bos

rdagger, 2019-10-23:
That's great!  I look forward to trying it out.  I linked the new repo on my video and website.
Thanks.

jczic, 2019-10-23:
Oh thank you!
I hope you will like it :)

Have a great day!