jczic/MicroWebSrv #8: Chrome not load all files ?

straga, 2018-01-04:
Why Chrome, Opera not loading all files (index.html. style.css, bootstrap.css, jquery.js, app.js). Every F5 load only some files?

FireFox works.

jczic, 2018-01-06:
Hi,
It's possible if MicroWebSrv process in no multi-thread mode.
Do you have try with multi-thread to True ?

straga, 2018-01-07:
It ESP8266.

jczic, 2018-01-07:
But it can accept one connection at the same time if it's not in multi-threading mode.

straga, 2018-01-07:
ESP8266 not support _thread, or you meaning something also ?

jczic, 2018-01-08:
I've never tested it on esp8266 but it works fine on ESP32 and Pycom modules.
It's strange that don't loads all files on your chrome but threads config is maybe the problem ?

straga, 2018-01-08:
hm ESP8266 not have threads. Need write other Http server for ESP8266.

closed 2018-01-08 by straga

straga, 2018-01-15:
I try write webserver for ESP8266 with timer without bloking. Runner run every 100ms for check connect. Load file by file.
https://www.youtube.com/edit?video_id=VdtNRpn-UfA