jczic/MicroWebSrv #10: POST and WebSocket does not work :-(

vrubel, 2018-01-27:
Examples from the delivery were used

On POST:
500 Internal Server Error
Server got itself in trouble 

On MicroWebSocket Test:
ERROR : undefined

-- DISCONNECTED --

Client - Firefox 57.0.4

P.S.
GET - OK
test.pyhtml - OK

jczic, 2018-01-27:
Hello,
What's your microcontroller ?
That's works on firefox for pycom and ESP32 normally.
It's maybe an out of memory internal error ? MicroWebSrv with all modules
requires a lot of memory.

vrubel, 2018-01-28:
I use [LOLIN32 Pro](https://wiki.wemos.cc/products:lolin32:lolin32_pro) and micropython from [loboris](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo)  

I do not think that the reason is lack of memory:  
```
>>> micropython.mem_info()
stack: 752 out of 19456
GC: total: 3201728, used: 13152, free: 3188576
 No. of 1-blocks: 162, 2-blocks: 27, max blk sz: 324, max free sz: 199222
>>> machine.heap_info()
Heap outside of MicroPython heap:
---------------------------------
              Free: 166892
         Allocated: 92040
      Minimum free: 166748
      Total blocks: 229
Largest free block: 113804
  Allocated blocks: 225
       Free blocks: 4

SPIRAM info:
------------
              Free: 913448
         Allocated: 3280672
      Minimum free: 910004
      Total blocks: 37
Largest free block: 912776
  Allocated blocks: 32
       Free blocks: 5
```

Liberasys, 2018-01-29:
Loboris has kindly updated MicroWebSrv after my request. See : https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/issues/39
Now websocket example is working with Loboris ESP32 firmware.
NB : websocket server in menuconfig is not the one that comes with MicroWebSrv.

vrubel, 2018-01-29:
I used MicroWebSrv code from your repository. (and not in menuconfig).
Thank you. I will try again. I'll build the last version of the micropython from Loboris.

vrubel, 2018-01-29:
I confirm.
Everything worked.
Thank you so much.

closed 2018-01-29 by vrubel

jczic, 2018-01-29:
Thank you guys :)