jczic/MicroWebSrv #11: Handling multipart/form-data requests

ayoy, 2018-02-04:
First of all thanks a lot for your excellent work on the HTTP server!

I'm wondering whether you support `multipart/form-data` requests. I'm trying to upload a file using curl:

    curl -v http://192.168.1.100/test -F "image=@test.txt"

but I'm getting an empty bytearray from a call to `ReadRequestContent()`. Am I doing something wrong or it's not supported yet?

jczic, 2018-02-05:
Hello and thank you !
Exactly, sorry but MicroWebSrv doesn't support multipart/form-data POST at
the moment :/
This subject stay again for future.

closed 2018-07-06 by jczic

nevercast, 2018-07-23:
Is this now implemented?

jczic, 2018-07-23:
nevercast : No for the moment ...
But you can easily use AjAX, or WebSockets to transfer datas :)

nevercast, 2018-07-23:
I've roughly implemented this myself.

```

        def _readRequestPostedMultipart(self, data, content_type_info):
            boundary = content_type_info['boundary']
            if isinstance(boundary, bytes):
                boundary = boundary.decode('utf8')
            if isinstance(data, bytes):
                data = data.decode('utf8')
            boundary_tokens = data.split(boundary)
            # First is empty, last is --
            boundary_tokens = boundary_tokens[1:-1]
            content = {}
            for token in boundary_tokens:
                # Starts with new line, strip that, 
                # Ends with two new lines, strip them
                token = token[2:-4]
                # Make a single split on the first new line.
                disposition, value = token.split('\r\n', 1)
                if len(value) > 2:
                    value = value[2:] # Excess leading new line.
                disposition_tokens = list(map(lambda clause: clause.strip(), disposition.split(';')))
                content_disposition = disposition_tokens[0]
                disposition_tokens = disposition_tokens[1:]
                for dis_tok in disposition_tokens:
                    field, field_value = dis_tok.split('=', 1)
                    field_value = field_value[1:-1]
                    if field == 'name':
                        content[field_value] = value
            return content
        # ------------------------------------------------------------------------
        def _readRequestPostedURLEncoded(self, data):
            if len(data) > 0:
                elements = data.decode().split('&')
                for s in elements:
                    param = s.split('=', 1)
                    if len(param) > 0:
                        value = MicroWebSrv._unquote_decode(
                            param[1]) if len(param) > 1 else ''
                        res[MicroWebSrv._unquote_decode(param[0])] = value
            return res
        # ------------------------------------------------------------------------

        def _parseContentType(self, contentTypeHeader):
            parts = list(map(lambda part: part.strip(), contentTypeHeader.split(';')))
            content_type = parts[0]
            content_type_info = {
                'type': content_type
            }
            if len(parts) > 1:
                for part in parts[1:]:
                    key, value = part.split('=', 1)
                    content_type_info[key] = value
            return content_type_info

        def ReadRequestPostedFormData(self):
            res = {}
            data = self.ReadRequestContent()
            print('ReadRequestPost')
            print(self._headers)
            print(data)
            if 'content-type' in self._headers:
                content_type = self._parseContentType(self._headers['content-type'])
                print(content_type)
                if content_type['type'] == 'multipart/form-data':
                    return self._readRequestPostedMultipart(data, content_type)
            return self._readRequestPostedURLEncoded(data)

        # ------------------------------------------------------------------------

```

It doesn't follow your style, is hastily written and has no tests. But I needed it working for an End of Sprint demo. I can make a PR possibly later this week. I've also added gzip "support" if the files are gzipped on flash.

jczic, 2019-10-20:
Hello,

I released a fully new version (v2.0) of my web server here :
github.com/jczic/MicroWebSrv2.
Open source MIT, fully asynchronous, more robust, more fast and more
efficient!
It is delivered with a good documentation.

Thank you for your support and feedback. ☺️

Jean-Christophe Bos