jczic/MicroWebSrv #12: MicroWebSrv not started when in AP mode

Liberasys, 2018-02-09:
Hello JC,
I have a new issue maybe in side effect of loboris network status improvements.
Your server code checks WLAN connection before starting. But in AP mode, the connection is not made, thus not starting the web server.

Maybe you could follow the example of lemariva in his FTP server ?
```

        self.wlan = network.WLAN(network.AP_IF)
        if self.wlan.active():
            ifconfig = self.wlan.ifconfig()
        else:
            self.wlan = network.WLAN(network.STA_IF)
            if self.wlan.active():
                ifconfig = self.wlan.ifconfig()
            else:
                dbg("No active connection")
```

jczic, 2018-02-10:
​The server code don't checks WLAN connection ..?
It doesn't this case in my src.​

Liberasys, 2018-02-12:
Ok, sorry, it is due to Boris modifications :
`       if not self._started :
            if not network.WLAN().isconnected():
                print("WLAN not connected!")
                return
`
I will see with him.
Sorry for inconvenience.
Thank you JC.

closed 2018-02-12 by Liberasys

jczic, 2018-02-12:
Ok no souci, it's possible to check WLAN configuration and not
"isconnected" :)