jczic/MicroWebSrv #19: Websockets do not work if you access them from the browser under iOS

vrubel, 2018-03-03:


jczic, 2018-03-03:
Hmm I'll test that... Strange :/

jkDesignDE, 2018-03-04:
Note : I have sometimes problems getting an answer from the websocket when the connection takes too long. The timeout is set to 500ms in JS at function onOpen(evt)

vrubel, 2018-03-04:
In this case, it is the different behavior in browsers on the PC and in iOS. On a PC all works, in iOS a connection break.

jczic, 2018-03-04:
Can you have a debugging information for iOS @vrubel ? I don't have an iPhone for the moment.

vrubel, 2018-03-05:
WebSocket connection to 'ws://192.168.1.45/' failed: Invalid HTTP version string: HTTP/1.0

vrubel, 2018-03-05:
`
diff --git a/microWebSrv.py b/microWebSrv.py
index 2f61358..c218a6f 100755
--- a/microWebSrv.py
+++ b/microWebSrv.py
@@ -545,7 +545,7 @@ class MicroWebSrv :
 
         def _writeFirstLine(self, code) :
             reason = self._responseCodes.get(code, ('Unknown reason', ))[0]
-            self._write("HTTP/1.0 %s %s\r\n" % (code, reason))
+            self._write("HTTP/1.1 %s %s\r\n" % (code, reason))
 
         # ------------------------------------------------------------------------
`
Does not solve the problem completely.
Now the connection setup for the websockets is not complete.

By the way, the effect is reproduced when using the PC version of the Safari browser.

jczic, 2018-03-05:
Ok thank you very mutch, I'll see that 😊

jczic, 2018-03-05:
Yes, I've updated the 1.0 to 1.1 version of HTTP and I've added a Content-Length header of 0 in the switching protocols headers... That's better for firefox ....!
But I don't found the solution for safari, it's very strange because safari seem waits an header after the handshake response.

vrubel, 2018-03-05:
I checked the safari work with the ws: //echo.websocket.org/ site.
Perhaps Wireshark will help to identify the difference in traffic.

jczic, 2018-03-05:
I've analysed too and websocket.org have the same code in they test :)
But there is a problem with the end of the headers I think ...
That's works anywhere (android, windows IE, Chrome Firefox, Mac Chrome,
Firefox) but not in safari.... I don't understand for the moment :'(

vrubel, 2018-03-05:
I captured the servers' answers. Perhaps this will help you in your search for a reason.

[image]

[bad.txt](https://github.com/jczic/MicroWebSrv/files/1780874/bad.txt)
[good.txt](https://github.com/jczic/MicroWebSrv/files/1780877/good.txt)

[bad.pcapng.zip](https://github.com/jczic/MicroWebSrv/files/1780879/bad.pcapng.zip)
[good.pcapng.zip](https://github.com/jczic/MicroWebSrv/files/1780881/good.pcapng.zip)

Maybe there is a problem with this?
[image]

jkDesignDE, 2018-03-05:
I assume that "Upgrade: websocket\r\n" have to be at last line

jczic, 2018-03-05:
Yes I've found :)
That's works now !
it's a bug with the base64 encoder...
I've pushed the fix ;)

jczic, 2018-03-05:
Thank you for this report 👍

closed 2018-03-05 by jczic