jczic/MicroWebSrv #21: Does it work on esp8266?

njXuanX, 2018-03-24:
A very good webserver for micropython, even with websocket and template!
My question is, does this powerful lightweight web server work on esp8266? After all, micropython offically support 8266, and 32 and 8266 using same basic MicroPython libraries. Or should I do some modify to make it works.
Thanks a lot!

jczic, 2018-03-26:
Hello and thank you :')
I have never tested it on esp8266 but do you can do it ?

njXuanX, 2018-03-27:
@jczic Seems dosent work, actually can not download the microWebSrv lib to 8266, I use uPycraft as IDE.
[image]

jczic, 2018-03-27:
Yes ok, you have not enought RAM memory to load MicroWebSrv... It works fine with modules who have 4 SDRam extension.

njXuanX, 2018-03-28:
@jczic Thanks a lot, still a very good project.

closed 2018-03-28 by jczic