jczic/MicroWebSrv #23: Core dump

SlowBro904, 2018-05-06:
What else do you need to assist troubleshooting this? MicroWebSrv was compiled into frozen bytecode and I'm running on a WiPy 2.0 with 512KB RAM and 4MB flash.

```import gc
from machine import Pin, info
from micropython import mem_info

gc.enable()
running = False

def web_srv_callback(web_srv_pin):
	''' If we hit the web_srv_pin launch the web server for configuration '''	
	# Debounce
	from utime import sleep, sleep_ms
	sleep(1)
	
	# Only call this function once
	global running
	if running:
		return
	
	from network import WLAN
	from microDNSSrv import MicroDNSSrv
	from microWebSrv import MicroWebSrv
	from microWebTemplate import MicroWebTemplate
	
	wlan = WLAN()
	
	MicroDNSSrv.Create({ '*' : '192.168.4.1' })
	
	if not web_srv_pin.value() == 0:
		sleep(1)
		return False
	
	# Give the clients 5 minutes to connect
	for i in range(300):
		if not wlan.isconnected():
			sleep(1)
		else:
			break
	
	if not wlan.isconnected():
		return False
	
	gc.collect()
	print('gc.mem_free(): ' + str(gc.mem_free()))
	print('machine.info(): ' + str(info()))
	print('gc.mem_alloc(): ' + str(gc.mem_alloc()))
	print('micropython.mem_info(): ' + str(mem_info()))
	
	mws = MicroWebSrv(bindIP='192.168.4.1', webPath="/flash/web_admin/")
	mws.SetNotFoundPageUrl("http://BetterAutomations.local/")
	mws.Start()
	sleep(1)
	running = True
	
	@MicroWebSrv.route('/')
	def handlerFuncGet(httpClient, httpResponse):
		print("In GET-TEST HTTP")
	
	@MicroWebSrv.route('/', 'POST')
	def handlerFuncPost(httpClient, httpResponse):
		print("In POST-TEST HTTP")

web_srv_pin = Pin('P6', mode=Pin.IN, pull=Pin.PULL_UP)
web_srv_pin.callback(trigger=Pin.IRQ_FALLING, handler=web_srv_callback)
```

```---------------------------------------------
System memory info (in bytes)
---------------------------------------------
[... 406 lines cut ...]
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=
RHb9P7B0/T88dv0/
sHT9P9B1/T81WeFC/Iv9P6xO/D9Edv0/pE78PwEAAAAUcv0/FHL9P0R2/T8Mcv0/
GAAAAEBy/T9pcGMxAErf3lr/NmBJ0zAAAQAAADx2/T8AAAAAIQAGABgAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAEf0/aBH9P9AR/T8AAAAAAAAAAAEAAAAAAAAA
11dAPwAAAAAUoQlAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAANm2HA==
xDIIQCSlCEAwAwYA/BwIgHB1/T/ocf0/AAAAADRy/T8AAAAA6HH9PwAAAAAkpQiA
UHX9PwEAAAD/////IFH8PwEAAAAEGfw/IAMGAAAAAAD//z+zAAAAAAAAAAAAAAAA
AAAAAL00CEABAAAASMAIQARw+j8AAAAAAAAAAAAAAAD//z+zAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAER2/T8BAAAA/gAAACMABgAAAAAAsHX9PwRO/D8AAAAA
AAAAAFoEAAAAAAAAIAMGAP////8AAAAABE78P/BN/D8QbP0/AAAAAAEAAAAAAAAA
AAAAANB1/T8AAAAAAAAAAPRKCEABAAAAAAAAAAAAAABhEgiAgH3+PwEAAAAAAAAA
AAAAAAAAAADcdf0/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAA
================= CORE DUMP END =================
Rebooting...
MicroPython 7c0b797 on 2018-05-05; WiPy with ESP32
Type "help()" for more information.
>>>
```

jczic, 2018-05-08:
It seem that you don't have enought memory to load microwebsrv ...
-> System free heap: 15860
A ESP32 with 4MB Flash or an OEM module of WiPy is required ... :-/
Not the dev WiPy Platform...

jczic, 2018-05-08:
Check Pycom OEM modules :
https://pycom.io/webshop#oem-products

SlowBro904, 2018-05-08:
Crap I was afraid you’d say that. Is it that your code requires the extra
ram always or is it just my other code crowding out the available ram? I
did try to put 100% of my code as frozen and that improved it slightly but
it still dumped.

closed 2018-06-09 by jczic