jczic/MicroWebSrv #30: html and js gzip support

Javierete, 2018-05-22:
Hello, 

First of all, congrats for the great project and for sharing!.

I'm trying to make MicroWebSrv serve gzipped index and js files (just testing doing an app with react). But at the moment can't make it work.

I've modified microwebsrv.py file adding:
`"index.html.gz"` under _indexPages
And
`".gz"    : "application/gzip"` under _mimeTypes

I'm really novice, i'll investigate further, just asking if someone has better knowledge to help me

jczic, 2018-05-22:
Hello,
If you want to serve .gz files, you can add directly the mime type to the list like that :
MicroWebSrv. _mimeTypes[".gz"] = "application/gzip"
That's works ?

Javierete, 2018-05-22:
Hello @jczic 
I've added gz mimeType but it downloads the file ("index.html.gz") instead of rendering the html page.
Tried also adding "index.html.gz" to _indexPages.

Maybe I'm not very clear before! Many thanks for your help!

[image]

jczic, 2018-05-25:
Unzip/GZ implementation doesn't exists on pycom modules.
It must be rewrited in mycropython or implemented in C on firmware directly ...

closed 2018-05-25 by jczic