jczic/MicroWebSrv #33: Basic Authentication Implementation

lmsoria, 2018-06-15:
Hi,
I'm trying to implement a Basic Authentication method when I access a web site allocated on a LoPy4.

I've successfully managed to do this (in a quite naive way) inside a route handler:

```
@MicroWebSrv.route('/auth')
def _httpHandlerAuth(httpClient, httpResponse) :
    responseCode = 401

    requestHeader = httpClient.GetRequestHeaders()
    print(requestHeader)

    try:
        rta = validateAuthentication(requestHeader['Authorization'])
        if(rta == True):
            responseCode = 200
    except KeyError:
        pass

    print("response Code: ", responseCode)
    if(responseCode == 200):
        myHeader = dict()
        myHeader['Access-Control-Allow-Origin'] = '*'
        myHeader['Access-Control-Allow-Methods'] = 'GET, POST'
        httpResponse.WriteResponseOk(headers		 = myHeader,
                                     contentType	 = "text/plain",
                                     contentCharset = "UTF-8",
                                     content 		 = 'Login OK :)' )
    else:
        myHeader = dict()
        myHeader['Access-Control-Allow-Origin'] = '*'
        myHeader['Access-Control-Allow-Methods'] = 'GET, POST'
        myHeader['Authorization'] = 'Basic'
        myHeader['WWW-Authenticate'] = 'Basic realm="401"'
        httpResponse.WriteResponse( code = responseCode,
                                    headers		 = myHeader,
                                    contentType	 = "text/plain",
                                    contentCharset = "UTF-8",
                                    content 		 = None)

def validateAuthentication(request):
    # Mock User and Password
    user = "admin"
    pwd = "1234"

    ret = False

    request_enc = request.split(" ")[1]
    request_dec = str( base64.b64decode(request_enc) )
    aux = request_dec.split("'")
    aux = aux[1].split(":")
    print("aux: ", aux)

    request_user = aux[0]
    request_pass = aux[1]
    print('User: ', request_user)
    print('Pass: ', request_pass)

    if(request_user == user and request_pass == pwd):
        print("OK!!!")
        ret = True
    return ret
```

I would like to implement something similar to this but before accessing the index.html file allocated inside the **/www** directory. If I write the same code as before, but using **@MicroWebSrv.route('/')** then I cannot access the index.html file (as expected).

Is there a way to implement this authentication method? Thanks in advance!

jczic, 2018-06-17:
Hello, can you try with *@MicroWebSrv.route('') ?*

lmsoria, 2018-06-18:
Hi @jczic, I forgot to mention that I've tried your proposal, and the results remains the same. It shows me the authentication dialog, and then a black page with "Login OK :)" message.

jczic, 2018-06-18:
If your handler function is triggered, the server doesn't return a page.
May be try to delete the real html index page (the file) and respond on the
fly ?

lmsoria, 2018-06-18:
I've managed to show the index page in two different ways.

The first one (not so aesthetic, at least for me) using ```httpResponse.WriteResponseRedirect('/index.html')``` if the validation is OK. Is simple but in my case shows then on the navigator ```device_IP/index.html#/``` and I particularly don't like how it looks.

The second one following your proposal:
```
f = open('/flash/www/index.html', 'r')
content= f.readall()
httpResponse.WriteResponseOk(headers = myHeader, contentType = "text/html", contentCharset = UTF-8", content = content)
```

Both ways worked fine, but I prefer the second one. Thank you for the suggestions!

jczic, 2018-06-18:
Yes, the second solution is better :)
You can also use .pyhtml files to code micropython in yours webpages
directly :)

closed 2018-07-02 by jczic

jczic, 2019-10-20:
Hello,

I released a fully new version (v2.0) of my web server here :
github.com/jczic/MicroWebSrv2.
Open source MIT, fully asynchronous, more robust, more fast and more
efficient!
It is delivered with a good documentation.

Thank you for your support and feedback. ☺️

Jean-Christophe Bos

Temik26, 2022-03-03:
Do you have a function(decorator) to implement basic auth?

Temik26, 2022-03-03:
Do you have a function (decorator) to implement basic auth?

Temik26, 2022-03-03:
Do you have a decorator to implement basic auth?