jczic/MicroWebSrv #36: Noob question - how to access 'pyhtml' pages?

ivandavidov, 2018-07-02:
Hello,

I'm trying the ``MicroWebSrv`` engine and it seems that serving static pages works fine. Using the controller methods also works fine. However, I have a serious issue with the ``pyhtml`` engine.

That's what I did:

* I copied ``microWebSocket.py``, ``microWebSrv.py`` and ``microWebTemplate.py`` in ``/flash/lib``
* I created simple static HTML page in ``/flash/www/index.pyhtml``
* I used the same initializiation code as from the provided sample

The initialization code:

```
from microWebSrv import MicroWebSrv
mws = MicroWebSrv()
mws.Start()
```

The issue is that whenever I try to access ``http://my-host``, my WiPy device hangs for a while and then it seems to restart itself. This happens only when I hit ``pyhtml`` pages. Like I said, there is no problem serving static HTML files and there is no problem serving pages from controller methods. I guess I'm missing something. Could you please explain what I might be missing in my setup?

Thanks!

ivandavidov, 2018-07-02:
Very interesting - when I start the server like this...

``mws.Start(threaded=False)``

...then my WiPi device doesn't restart when I hit ``pyhtml`` pages. Instead, I get HTTP error 501 (Not Implemented).

jczic, 2018-07-02:
Ok, please, update the Git repository or remove the '.' from the 2 import lines in microWebSrv.py, at the beginning, and tell me.
I've just changed this problem now, this syntax doesn't works on LoPy/WiPy.
Thank you

jczic, 2018-07-02:
You can try : http://x.x.x.x/test.pythml for PyHTML "language"
but also : http://x.x.x.x/wstest.html for testing websockets :)

ivandavidov, 2018-07-02:
OK, seems there is some progress. My WiPy device no longer restarts when I hit static HTML inside 'pyhtml' file, but when I try to open [test.pyhtml](https://github.com/jczic/MicroWebSrv/blob/master/www/test.pyhtml), I get this:

```
PyHTML page execution error
maximum recursion depth exceeded
```

ivandavidov, 2018-07-02:
This is my custom pyhtml which also throws the same 'maximum recursion depth exceeded' error:

```
<html>
  <head>
    <title>TEST PYHTML</title>
  </head>
  <body>
    <h1>BEGIN {{ "Hello" }} END</h1>
    <ul>
      {{ for test in range(5) }}
        <li>{{ test }}</li>
      {{ end }}
    </ul>
  </body>
</html>
```

ivandavidov, 2018-07-02:
I updated the firmware earlier today, so updating the firmware is not an option until new firmware comes out. Regarding the MicroPyhon options - I have no idea what you're talking about but I'll dig it further. I admit I'm still quite new to both WiPy and MicroPython.

For now I'm more than happy with the ability to process HTTP requests by using controller methods and that's enough for my use cases. Thank you for the assistance on this matter!

jczic, 2018-07-02:
Cool and thank you :)

Just note that WiPy/LoPy development modules have a very little RAM memory compared to OEM versions.
If you want the threaded option of MicroWebSrv for example, I think that you doesn't have enought memory...

closed 2018-07-02 by jczic

ivandavidov, 2018-07-02:
FYI - I rebooted my device and suddenly my sample 'pyhtml' page worked. :)

I guess there has been a lot of garbage in the RAM and after reboot it cleared. It's strange, considering I should have approximately 4MB RAM according to the specs.

Anyway, problem solved.

jczic, 2018-07-02:
Yeah 🥇

jczic, 2019-10-20:
Hello,

I released a fully new version (v2.0) of my web server here :
github.com/jczic/MicroWebSrv2.
Open source MIT, fully asynchronous, more robust, more fast and more
efficient!
It is delivered with a good documentation.

Thank you for your support and feedback. ☺️

Jean-Christophe Bos