jczic/MicroWebSrv #37: Out of memory on main.py startup

nobody102, 2018-07-04:
Running micropython on nodemcu.   Copied MicroWebServ.py and main.py using webrepl.
Upon restart of the board

Traceback (most recent call last):
  File "main.py", line 2, in <module>
MemoryError: memory allocation failed, allocating %u bytes

MicroPython v1.9.4-8-ga9a3caad0 on 2018-05-11; ESP module with ESP82

Any suggestions? THanks

nevercast, 2018-07-04:
The ESP8266 only has about 80KB of memory available before the Micropython Runtime is added, if you open a reply and do 
```
import gc
gc.mem_free()
```
You should be able to see how much memory is available, my feeling is that its going to be very little. MicroWebSrv is very full featured and as such is going to have a large memory footprint. I'm using it on a Pycom LoPy4 which has about 2MB of memory (as it features an external SPI RAM).

You can try either using a .mpy file or by baking the MicroWebServ.py module in to ROM by freezing it during compile. Both will lower the load on working memory.

The first method, you will need to use mpy-cross, which takes a python file and compiles it to a binary format. Usually MicroPython does this at run time, but it needs enough RAM to load the raw .py file as well as the resulting binary. Your module during import is essentially in RAM twice during the compile step. By using mpy-cross on your machine and only uploading that file, you skip the compile step, and instead the .mpy file is loaded in to memory and run direction (not compiled on the NodeMCU)

The second method is freezing the module in to the firmware, which means compiling the firmware for NodeMCU yourself and giving it the MicroWebServ.py file to freeze. This will run mpy-cross, and burn this file in to the ROM. It will be read from the ROM at runtime and takes up 0 space in the RAM. The module will still use RAM for any dynamically allocated memory (strings, arrays, the usual python objects) but it will not occupy RAM for the code.

The last option which isn't really a solution, is don't use the NodeMCU or any 8266 board. The repo says ESP32 so I suggest using an ESP32 board. The Pycom devices (WiPy3) would be a good start there.

nobody102, 2018-07-04:
I am not wed to the nodemcu board, so will look at esp32 models.  Thanks

nevercast, 2018-07-05:
Hello again @nobody102.

If you've decided to go with an ESP32 board instead, would you agree this issue is closed?

jczic, 2018-07-06:
Oky, so I'm closing this issue

closed 2018-07-06 by jczic