jczic/MicroWebSrv #38: str.encode() changing contentLength in writeResponse()

FlyingYeti, 2018-07-07:
I'm using `writeResponseOK()` (I could have used `writeResponseJSONOK()` but nvm) to send a JSON to my app.
Though I get an error on app side (`JSON.parse()` failure) when my JSON contains the 'é' character (the only 'exotic' character I send). I found out that i had one character stripped from my response for each 'é'.
I looked at the code of MicroWebSrv.py and found that the `data.encode()` (in `_write()` line 578) is computed after `len(content)` (in `WriteResponse()` line 644). So in my use case I get `data.encode() - len(content) == n` where `n` is the number of 'é' and therefore as many character are stripped at the end of my content.
The fix is to compute  the `encode()` before the `len()`, maybe `contentLength = len(content.encode()) if content else 0` (and that's what I do outside of  `writeResponseOK()` as a quick fix)

`s = '\xe9'`
`print(s) # é`
`s.encode() # b'\xc3\xa9'`
`len(s) == len(s.encode()) # False`

jczic, 2018-07-14:
@FlyingYeti Ok yah, I'll check that as soon !
Thank you :+1:

jczic, 2018-07-23:
@FlyingYeti Sorry for the delay.
I've fixed it in commit https://github.com/jczic/MicroWebSrv/commit/7d756e8cb0e03f10aba354959ccaefa8763e645d
👍

closed 2018-07-23 by jczic