jczic/MicroWebSrv #39: Separe port/services for Websocket and Webserver

enrutador, 2018-07-23:
Hi Christophe 
Congratulations for your work ¡¡¡. I have a question about the constructor:

Is it possible to change the ports of the websocket process and the webserver process?
For example, the webserver works on port 80 and websocket on 81. I notice that while I have port 80 opened by ws: // the protocol http: // the web server stops working
A lot of thanks

jczic, 2018-07-23:
Hello and thank you !
You can change the port of the web server by passing **port=xxx** to the constructor.
WebSockets are handled by the web server, so this is the same port number.
If you want to have a web server (for get/post/ajax...) on a port and websockets on a different port, you must to instanciate 2 web server with different args and different management.
Another idea ? :)

closed 2018-07-23 by jczic

jczic, 2019-10-20:
Hello,

I released a fully new version (v2.0) of my web server here :
github.com/jczic/MicroWebSrv2.
Open source MIT, fully asynchronous, more robust, more fast and more
efficient!
It is delivered with a good documentation.

Thank you for your support and feedback. ☺️

Jean-Christophe Bos