jczic/MicroWebSrv #41: add support for list besides from string format of html file

curlyz, 2018-08-17:
For many case , html file will be loaded from flash into a string and then will be pass to WriteResponseOK.
In my application , having upto 50K of heap , but it is fragmented , a html file size of 6KB is failed to loads due to MemoryError
To overcome this , I create a list , load the html in chunk of 100bytes , modified the _write() to accept the  list , it write to socket pieces by pieces.
Since micropython doesnt handle heap fragmentation , I think my experiences with this issue will help .

jczic, 2018-08-17:
You are on a very limited microcontroller right ?
The memory of ESP32 and pycom OEM modules is around of 4Mb.
If you have a large .pyhtml file, how you solve it ?

curlyz, 2018-08-17:
I am using ESP32 for this , if the file is too big , then I would suggest 

```
def _httpResponse_GET():
  f = open('somthing.pyhtml')
  while True : 
    temp = f.read(100) 
    if len(temp) == 0: 
      break
    socket.write(temp)
  f.close()

```
I am not using readline because somehow , Python having problem with '\r' and '\n' , and also , I don't want to have problem with compressed file

jczic, 2018-08-17:
It's impossible with pyhtml ...

curlyz, 2018-08-17:
I see , since the filename can be passed through , the function simply check if the file is html or pyhtml
My method is just a error-prone method for large string file and I think it only work with html.

jczic, 2019-10-20:
Hello,

I released a fully new version (v2.0) of my web server here :
github.com/jczic/MicroWebSrv2.
Open source MIT, fully asynchronous, more robust, more fast and more
efficient!
It is delivered with a good documentation.

Thank you for your support and feedback. ☺️

Jean-Christophe Bos