jczic/MicroWebSrv #42: response too slow

Fuermohao, 2018-08-23:
thanks for sharing this amazing project!
the average of http-responsing time is too slow, about 3 seconds.
Is there a way to optimize?

jczic, 2018-08-26:
What's your platform and how memory SDRAM have you ?

Fuermohao, 2018-08-26:
esp32 , about 128k RAM

jczic, 2018-08-26:
There are many memory allocation in the web server, 128k is can be just if you have microwebsrv (websockets? pyhtml?) + another code around.

Do you use the threaded mode or the blocking mode? 3 seconds, it's slow yes...

You can also adjust threads stack size of your micro-controller.

However, this is not due to the DNS resolving?

closed 2018-09-07 by jczic

Fuermohao, 2018-09-07:
hi，buddy, I just bought a new esp32 chip with psram. And I'm going to test it tomorrow.

jczic, 2018-09-07:
Ok