jczic/MicroWebSrv #43: how to send updated json data from microwebserver to json file that embedded in html file

m2030, 2018-09-07:
great project  it is easy and simple 
how to send updated json data from microwebserver to json file that embedded in html file with jquery ajax 

`<!DOCTYPE HTML>
<html>
<head>
<script type="text/javascript">
window.onload = function() {
	var dataPoints = [];
	var chart;
	$.getJSON("/livegraph2.json", function(data) { 
	x=0; 

		$.each(data, function(key, value){
			dataPoints.push({x:x},{y: parseInt(value)});
			x++;
			xaxis=x;
		});
		chart = new CanvasJS.Chart("chartContainer",{
			title:{
				text:"Live graph for spo2 pleth"
			},
 axisX:{
         title: "time in microsecond"//,
       //  minimum: xaxis  , // change here
        // maximum: xaxis+100
     },
      axisY:{
         title: "in melevolt"//,
       //  minimum: xaxis  , // change here
        // maximum: xaxis+100
     },
			data: [{
				type: "line",
				dataPoints : dataPoints,
			}]
		});
		chart.render();
		updateChart();
	});
	var dataLength = 20;
	function updateChart() {
	$.getJSON("/livegraph2.json", function(data) {
		$.each(data, function(key, value) {
			dataPoints.push({
			x:x,
			y: parseInt(value)
			})
			x++;
	if (dataPoints.length > dataLength) {
		dataPoints.shift();
	}
		});

		chart.render();

		setTimeout(function(){updateChart()}, 1000);
	});
	}
}
</script>
<script type="text/javascript" src="https://canvasjs.com/assets/script/jquery-1.11.1.min.js"></script>
<script type="text/javascript" src="https://canvasjs.com/assets/script/canvasjs.min.js"></script>
</head>
<body>
<div id="chartContainer" style="height: 300px; width: 100%;"></div>
</body>
</html>`
like with micropython code `
import gc
import os 
gc.collect()
from microWebSrv import MicroWebSrv
array=[80,90,86,89,90,89,86,95,94,91,89,97,94,100,110]
array1='['
def _httpHandlerarray(httpClient, httpResponse):
  gc.collect()
  for x in array:
    array1=str(x+7)
    array1=','
  array1[-1]=']'
## f=open('/www/livegraph2.json','w')
 ## f.write(mystring)
 ##f.close()  
 ##print(mystring)
  httpResponse.WriteResponseOk(headers=None, 
  contentType= "application/json", 
  contentCharset="UTF-8", 
  content=array1)      
      
routeHandlers=[("/livegraph2.html","GET",_httpHandlerarray)]
srv=MicroWebSrv(routeHandlers=routeHandlers,webPath='/www/')
gc.collect()
print(array1)
srv.Start(threaded=False)`

jczic, 2018-09-17:
Hello :)
Sorry for latence !

You can just respond in JSON format from a handler function like that (check main.py) :
```
    httpResponse.WriteResponseJSONOk({
        'example1'    : 12345,
        'example2'    : True,
        'example3'    : 'toto'
    })
```

m2030, 2018-09-18:
thanks jczic

closed 2018-09-18 by m2030