jczic/MicroWebSrv #44: nmap scan stops MicroWebSrv

dl2080, 2018-09-26:
Hello,
thank you for this nice web server project!

I'm using _nmap_ to discover devices in my network. However, as soon as I start the scan in the linux console, MicroWebSrv on my WiPy 3.0 stops working. After that, restarting the server with `srv.Start()` fails due to _[Errno 12] ENOMEM_. I found that I need to close the socket first. `srv.Stop()` will not help, because it closes only, if `srv._started` is true, which is not the case. Maybe this can be changed? `srv._server.close()` helps. After that a restart of the server is possible without reseting the whole device.

To trace the cause of the server stop error, I changed the try/except block in microWebSrv.py line 239 to
```
try :
   client, cliAddr = self._server.accept()
except Exception as e :
   print(str(e))
   break
```
I get _[Errno 128] ENOTCONN_ when _nmap_ starts.

It seems some connection is being lost. Is there a way to prevent MicroWebSrv from stopping at this position?

nmap has version 7.60
wipy firmware is 1.19.0.b4
micropython has version 1.9.4
Server is started with:
```
srv = MicroWebSrv(routeHandlers = routeHandlers)
srv.MaxWebSocketRecvLen = 256
srv.WebSocketThreaded = True
srv.AcceptWebSocketCallback = _acceptWebSocketCallback
srv.Start(threaded = True)
```

jczic, 2018-10-11:
Hello,
Yes I see that ! I've just updated the code to close server only if the error code is 113 (normal closing socket).
You can check at : https://github.com/jczic/MicroWebSrv/commit/5074d96203d32c049b66577013cad90ea8e21327
That's ok for you ?
Thanks !

dl2080, 2018-10-11:
Yes, it doesn't crash anymore!
Great! Thanks a lot!

closed 2018-10-11 by dl2080

jczic, 2019-10-20:
Hello,

I released a fully new version (v2.0) of my web server here :
github.com/jczic/MicroWebSrv2.
Open source MIT, fully asynchronous, more robust, more fast and more
efficient!
It is delivered with a good documentation.

Thank you for your support and feedback. ☺️

Jean-Christophe Bos