jczic/MicroWebSrv #45: always got the 404 error

dukeduck1984, 2018-10-03:
Hi,

I always got the "404 Not Found  Nothing matches the given URI" error, although I could connect to the server. 

The similar issue was reported here too: https://www.cnx-software.com/2017/10/16/esp32-micropython-tutorials/

I used the loboris fork of the Micropython firmware for ESP32: https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/.  The MicroWebSrv module was frozen in the firmware.

Here is my code:
`
from microWebSrv import MicroWebSrv

ws = MicroWebSrv()

@MicroWebSrv.route('/')
def handlerFuncGet(httpClient, httpResponse):
    content = """\
    <!DOCTYPE html>
    <html>
        <h1>hello world</h1>
    </html>
    """
    print(content)

@MicroWebSrv.route('/<action>')
def handlerFuncEdit(httpClient, httpResponse, routeArgs):
    action = routeArgs['action']
    print(action)

ws.Start()
`

Thanks in advance for your help.

-Kaiyuan

formica-multiuso, 2018-10-08:
Hi have the exact same issue.
Any idea?

jczic, 2018-10-08:
Hello, root folder is "/flash/www" by default.
You can change it in the constructor of MicroWebSrv.
This is maybe the problem ?

dukeduck1984, 2018-10-09:
Hi,

Thanks for the reply.

So I will need the 'www' folder under flash root even it's empty?

-Kaiyuan

jczic, 2018-10-09:
Oh no sorry, your problem isn't the root folder.
Just, you must call a function to write the content, see httpResponse.WriteResponseOk() or others in the readme page.

dukeduck1984, 2018-10-11:
Ah!  I got you!  I should use httpResponse.WriteResponseOk() instead of print(), correct?

Thanks for pointing this out.

-Kaiyuan

jczic, 2018-10-11:
Yes !

closed 2018-10-11 by jczic

jczic, 2019-10-20:
Hello,

I released a fully new version (v2.0) of my web server here :
github.com/jczic/MicroWebSrv2.
Open source MIT, fully asynchronous, more robust, more fast and more
efficient!
It is delivered with a good documentation.

Thank you for your support and feedback. ☺️

Jean-Christophe Bos