jczic/MicroWebSrv #46: threaded=True is slow

silverchen0402, 2018-10-29:
Hi

I use your great module for my nodemcu-32s v1.1 board, micropython version 1.9.4, everything looks fine, but when I start webserver with threaded=True, the response is slow, about 4s, however it's ~200ms when threaded mode is False.

I have no idea why this happens, my micropython is clean, no other modules and I use IP to connect the device, so it seems not an DNS issue, could you give me some hints about this?

Thanks a lot.
-Silver

jczic, 2018-12-09:
Hello and sorry for my very late answer!
It is strange indeed..
Just, can you try on your platform to increase the stack size before creating the threaded server with a call to _thread.stack_size([size]) ?

jczic, 2018-12-09:
microWebSrv using just one synchronous pipe-line (threaded or not) and more for websockets connections.
I'll remake it with my async lib : https://github.com/jczic/XAsyncSockets :)

gengshenghong, 2018-12-19:
Thank you for your nice work.

Same issue here, it takes 3 seconds to respond with very simple http server (same with https://github.com/jczic/MicroWebSrv/issues/42). I can confirm it works with "mws.Start(threaded=False)"

Suggestion:
Document it clearly about the limitation or by default set it as False at least. For first-time users, this waste some time as we usually use "default configuration" for testing purpose, and I started to think what is wrong with my network (as I cannot imagine a http test request takes 3 secs!). Of course, if it can be root-cased and fixed, that is better. :) Looking forward to the synchronous version.

Thanks,
Shenghong

jczic, 2018-12-19:
Ok thanks.
I will test further because it doesn't make sense for a single thread on the server to take so long.

jczic, 2018-12-19:
So, I have removed the thread allocation testing :
https://github.com/jczic/MicroWebSrv/commit/73cd549f926d2bd1c9a60e9979a53283d547ee15
But on computer or my emmbeded controller, the thread mode is faster than the no-threaded.
Strange ?

gengshenghong, 2018-12-21:
@jczic It seems behavior different on my esp32 WROOM 32 D and Lyrat (ESP32 WROVER + psRam). Behavior is totally strange.
(1) on WROOM, with or without threaded, the response time is 2.2 secs.
(2) on Lyrat, with thread, 3 seconds, without threads, 100ms. And at the same time, I tried to test by adding some more threads which just sleep, on Lyrat, it will affect the response time of http very much (more threads, slower, even 8 secs with 10 threads), and in WROOM it is not affected.
 
I known Lyrat has psRam which may affect performance and I known there is a special version supporting it (I am not using it now, I just use normal micropython build).

My target is to use normal WROOM module without psRam, it seems like performance is not affected by adding more sleep threads, but the response time is longer as 2.2 seconds with or without threads mode. I may need to focus on WROOM for more testing now. Can you provide the board information you are using for testing? Which board is the minimal requirements for a simple http server?

Thanks,
Shenghong

gengshenghong, 2018-12-21:
@jczic some updates, please take a look at https://github.com/micropython/micropython/issues/4372

Now I focus on esp32 WROOM (no psRAM) module and take some careful tests.
I got same results using MicroWebSrv for the testing. It seems like thread behavior is strange in micropython....

jczic, 2019-02-03:
Hello @gengshenghong, some news about this problem with readline IO ?

closed 2019-02-28 by jczic

jczic, 2019-10-20:
> micropython/micropython#4372

Hello,

I released a fully new version (v2.0) of my web server here :
github.com/jczic/MicroWebSrv2.
Open source MIT, fully asynchronous, more robust, more fast and more
efficient!
It is delivered with a good documentation.

Thank you for your support and feedback. ☺️

Jean-Christophe Bos