jczic/MicroWebSrv #47: noob question- examples not working.

nedoskiv, 2019-01-01:
Hello,
I was switching to micropython and start to looking for web server, found this wonderful server and decided to test it. I was able to start the server from basic example and access root "/" also open PDF example file, but when decided to test .pyhtml it returns an error (http://192.168.7.96/test.pyhtml):

PyHTML page execution error
unexpected indent (line 1) 

another error seen when try (http://192.168.7.96/wstest.html):

MicroWebSocket Test :

Connection to ws://192.168.7.96...

ERROR : undefined

-- DISCONNECTED --

perhaps I'm missing something, my python knowledge is far from good :)
I'm using micropython 1.9.4 on ESP32

jczic, 2019-01-01:
Hello Nedoskiv,
Hmm very strange in effect!
I will try to check on a 1.9.4 micropython but it should absolutely not have this kind of websockets problem or indentation problem.
The examples also work in standard python mode if you do "import main" and go to http://localhost/test.pyhtml or http://localhost/wstest.html.
People use it on ESP32 (VROOM for example) and that's works.

nedoskiv, 2019-01-01:
thank you for quick response, in my test I do not use import main,, when i do that it going worse:

[0;32mI (125668) modsocket: Initializing[0m
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400f22cb  PS      : 0x00060830  A0      : 0x800df8e4  A1      : 0x3ffb6680  
A2      : 0x3ffeaf50  A3      : 0x3ffeaff0  A4      : 0x00000000  A5      : 0x3ffeab20  
A6      : 0x00001292  A7      : 0x00000002  A8      : 0x00000000  A9      : 0x3ffb6650  
A10     : 0x00000038  A11     : 0x3ffb6680  A12     : 0x00000002  A13     : 0x00000002  
A14     : 0x00000001  A15     : 0x00001292  SAR     : 0x00000002  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000010  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x400f22cb:0x3ffb6680 0x400df8e1:0x3ffb66b0 0x400dbd01:0x3ffb66d0 0x400dbd69:0x3ffb66f0 0x400e8ea5:0x3ffb6710 0x400df9fc:0x3ffb67b0 0x400dbd01:0x3ffb6820 0x400dbd69:0x3ffb6840 0x400e8ea5:0x3ffb6860 0x400df9fc:0x3ffb6900 0x400dbd01:0x3ffb6970 0x400dbd2e:0x3ffb6990 0x400dc7ab:0x3ffb69b0 0x400e6831:0x3ffb6a40 0x400e6851:0x3ffb6a70 0x400df97a:0x3ffb6a90 0x400dbd01:0x3ffb6ac0 0x400e8e19:0x3ffb6ae0 0x400df9fc:0x3ffb6b80 0x400dbd01:0x3ffb6bf0 0x400dbd2e:0x3ffb6c10 0x400f8eba:0x3ffb6c30 0x400f9104:0x3ffb6cd0 0x400efc50:0x3ffb6d10

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4732
load:0x40078000,len:7496
load:0x40080400,len:5512
entry 0x4008114c
[0;32mI (399) cpu_start: Pro cpu up.[0m
[0;32mI (399) cpu_start: Single core mode[0m
[0;32mI (399) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (403) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (409) heap_init: At 3FFC4F68 len 0001B098 (108 KiB): DRAM[0m
[0;32mI (415) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (422) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (428) heap_init: At 40091604 len 0000E9FC (58 KiB): IRAM[0m
[0;32mI (434) cpu_start: Pro cpu start user code[0m
[0;32mI (5) cpu_start: Starting scheduler on PRO CPU.[0m
OSError: [Errno 2] ENOENT

in my test I simply do that:

from microWebSrv import MicroWebSrv
mws = MicroWebSrv()
mws.Start()

and set web path to proper location.

jczic, 2019-02-03:
Hello @nedoskiv, some news ?

nedoskiv, 2019-02-04:
Hello, I give up trying to fix something that I do not understand, instead of it write my own simple web server that run in _thread and consume low memory. So far so good, it servers my project needs.
If you wanna share my code, let me know

jczic, 2019-02-04:
What's your code ?

nedoskiv, 2019-02-05:
```
try:
  import usocket as socket
except:
  import socket
from machine import Pin
import uselect
import _thread
import time
import esp
esp.osdebug(None)
import gc
gc.collect()
led = Pin(2, Pin.OUT)
def web_page():
  if led.value() == 1:
    gpio_state="ON"
  else:
    gpio_state="OFF"
  
  html = """<html><head> <title>ESP Web Server</title> <meta name="viewport" content="width=device-width, initial-scale=1">
  <link rel="icon" href="data:,"> <style>html{font-family: Helvetica; display:inline-block; margin: 0px auto; text-align: center;}
  h1{color: #0F3376; padding: 2vh;}p{font-size: 1.5rem;}.button{display: inline-block; background-color: #e7bd3b; border: none; 
  border-radius: 4px; color: white; padding: 16px 40px; text-decoration: none; font-size: 30px; margin: 2px; cursor: pointer;}
  .button2{background-color: #4286f4;}</style></head><body> <h1>ESP Web Server</h1> 
  <p>GPIO state: <strong>""" + gpio_state + """</strong></p><p><a href="/?led=on"><button class="button">ON</button></a></p>
  <p><a href="/?led=off"><button class="button button2">OFF</button></a></p></body></html>"""
  return html

def ch(conn):
  print ("thread before recv")
  global thr
  poller = uselect.poll()
  poller.register(conn, uselect.POLLIN)
  res = poller.poll(100000)  # time in milliseconds
  print ("res",res)
  if not res:
    conn.close()
    print('timeout, close conn')
    thr -= 1
    return
  request = conn.recv(1024)
  request = str(request)
  print('Content = %s' % request)
  led_on = request.find('/?led=on')
  led_off = request.find('/?led=off')
  if led_on == 6:
    print('LED ON')
    led.value(1)
  if led_off == 6:
    print('LED OFF')
    led.value(0)
  response = web_page()
  conn.send(response)
  conn.close()
  thr -= 1
s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
s.bind(('', 80))
s.listen(5)
thr=0
while True:
  gc.collect()
  print (gc.mem_free())
  print ("before accept --> " + str(thr))
  if thr >2:
    time.sleep(1)
    print ("Thread limit reached, quening new connections")
    continue
  conn, addr = s.accept()
 # conn.send("ESP32 VEN's testing code")
  print('Got a connection from %s' % str(addr))
  thr += 1
  _thread.start_new_thread(ch, (conn,))
  print ("raised new thread, total thr -->" + str(thr))

```

This is an standalone example of something i found on web, just modified to serve new connections in different _threads and to have timeout on incoming connection. My current work is more complicated than that, but that can be a basic to those who want to have own web server

jczic, 2019-02-05:
Ok :)
So, in this case, you can check my another lib for make your own Web server, directly with asynchronous I/O :
https://github.com/jczic/xAsyncSockets

mrwhy-orig, 2019-02-16:
I can confirm the identiation error with the test.pyhtml on an ESP32 with  Micropython v1.10-54-g43a894fb4
The Websocket example is working fine.

jczic, 2019-10-20:
Hello,

I released a fully new version (v2.0) of my web server here :
github.com/jczic/MicroWebSrv2.
Open source MIT, fully asynchronous, more robust, more fast and more
efficient!
It is delivered with a good documentation.

Thank you for your support and feedback. ☺️

Jean-Christophe Bos

nedoskiv, 2019-10-21:
Thank you, I took a look at it, looks promising.

closed 2020-03-05 by jczic