jczic/MicroWebSrv #48: Curious whether MicroWebSrv supports https?

tshead2, 2019-01-26:
I’ve been wanting to handle web requests using https on a PyCom WiPy 3.0 for awhile now without success.  Maybe this is asking too much of the hardware?

Thanks in advance,
Tim

jczic, 2019-01-26:
SSL/TLS (https) isn't implemented on MicroWebSrv, but it's simple with a pycom.
Just wrap socket with the goods parameters after socket creation :)

tshead, 2019-01-28:
@jczic - Thanks for the response!  I thought that this would be easy too, but I've tried calling ssl.wrap_socket(...) after both bind() and accept(), and in both cases the client hangs while making a request.  This was with my own minimalist http server, but I think I'll try hacking on MicroWebSrv instead.  I'm curious if you have a working example?

Cheers,
Tim

tshead, 2019-01-29:
Thanks for all the help, I’ll check it out!

Cheers,
Tim

jczic, 2019-10-20:
Hello,

I released a fully new version (v2.0) of my web server here :
github.com/jczic/MicroWebSrv2.
Open source MIT, fully asynchronous, more robust, more fast and more
efficient!
It is delivered with a good documentation.

Thank you for your support and feedback. ☺️

Jean-Christophe Bos

tshead, 2019-10-20:
Thanks for the followup on this … I will check the new version out!

Cheers,
Tim