jczic/MicroWebSrv #50: Feature - Fine tune control over processing

washcycle, 2019-02-03:
I was thinking about creating a pull request to take the _serverProcess function and the ability to run the process outside in a while loop at the parent thread process.

    def _serverProcess(self) :
        self._started = True
        while True :
            try :
                client, cliAddr = self._server.accept()
            except Exception as ex :
                if ex.args and ex.args[0] == 113 :
                    break
                continue
            self._client(self, client, cliAddr)
        self._started = False

**Proposed Addition**

    def serverProcessOneCycle(self) :
        try :
            client, cliAddr = self._server.accept()
        except Exception as ex :
            if ex.args and ex.args[0] == 113 :
                return
        self._client(self, client, cliAddr)

**Example Usage**

```
from microWebSrv import MicroWebSrv
mws = MicroWebSrv()                                    # TCP port 80 and files in /flash/www
mws.MaxWebSocketRecvLen     = 256                      # Default is set to 1024
mws.WebSocketThreaded       = False                    # WebSockets without new threads
mws.AcceptWebSocketCallback = _acceptWebSocketCallback # Function to receive WebSockets

mws.Init() ## not sure about this first init part

def some_process():
    #something awesome

whilte True:
    some_process()
    try:
        mws.serverProcessOneCycle()
   except:
       print("something went wrong")

```

Any thoughts, perhaps I didn't consider an already present micropythonic way to do this.

washcycle, 2019-02-04:
> Hi,
> If you want to do that, without thread, you just can stop the server in
> the client connection callback event, no ? :)

Is there any excessive overhead if I do that? Otherwise that sounds like an
excellent idea, didn't even think of it. Thanks.

jczic, 2019-02-04:
No, because in any case, this is in the processing of the reception of the client socket that allows to close the server properly. The loop then comes out immediately (err #113) :)
No problem.

closed 2019-02-04 by jczic

jczic, 2019-10-20:
Hello,

I released a fully new version (v2.0) of my web server here :
github.com/jczic/MicroWebSrv2.
Open source MIT, fully asynchronous, more robust, more fast and more
efficient!
It is delivered with a good documentation.

Thank you for your support and feedback. ☺️

Jean-Christophe Bos