jczic/MicroWebSrv #52: Handle file uploads?

antonvh, 2019-02-18:
From the readme and I can not deduce how to write a file upload handler with microwebsrv. 

I currently use python3/tornado where I go: 
```
class UploadHandler(tornado.web.RequestHandler):
    def post(self):
        if 'file_0' in self.request.files:
```
etc.

I want to move to micropython for this web app. How do I do that?

jczic, 2019-02-28:
Hi, no, MicroWebSrv doesn't integrate files POST with forms data for the moment.

closed 2019-03-15 by jczic

jczic, 2019-10-20:
Hello,

I released a fully new version (v2.0) of my web server here :
github.com/jczic/MicroWebSrv2.
Open source MIT, fully asynchronous, more robust, more fast and more
efficient!
It is delivered with a good documentation.

Thank you for your support and feedback. ☺️

Jean-Christophe Bos