jczic/MicroWebSrv #53: Building firmware image, including this library, issue on pycom

jager07, 2019-02-21:
I successfully use the POST/GET services from your libraries on a FiPy board (ESP32) when it's uploaded the normal way. It works great, loads webpages from SD, allows me to configure settings in the device and save them back to a config file, all good.
The problem is when I compile the 3 libraries along with the fipy firmware (i've done this successfully many times) I get a check_sum error and the FiPy won't start. I remove the 3 library files and all good, it works again.
Maybe a problem with the compiler and how it 'freezes' .py files, I just don't know why this is happening. For no I will try and load these libraries on the SD card as a work around.

jager07, 2019-02-21:
Note it could also be a partition size problem, just realised this. Maybe I try a simple program with the websrv and see if that compiles ok. The FiPy is meant to have 8MB Flash, and my program is 1.6MB so should be no issue...

jager07, 2019-02-21:
It was the partition problem, I've extended the partition and it works. So your code is all good!

Thank you

closed 2019-02-21 by jager07

jczic, 2019-02-21:
👍 
Great ;)

jczic, 2019-10-20:
Hello,

I released a fully new version (v2.0) of my web server here :
github.com/jczic/MicroWebSrv2.
Open source MIT, fully asynchronous, more robust, more fast and more
efficient!
It is delivered with a good documentation.

Thank you for your support and feedback. ☺️

Jean-Christophe Bos