jczic/MicroWebSrv #54: Recursion error from .pyhtml when threaded=True

elixx, 2019-05-13:
If I start the process with  mws.Start(threaded=True), attempting to load any .pyhtml page renders:

```
PyHTML page execution error
maximum recursion depth exceeded (line 18)
```

This does not happen at all if I start with threaded=False.
Is this to be expected?

elixx, 2019-05-14:
I reviewed https://github.com/jczic/MicroWebSrv/issues/36  and tried a few different firmware versions, but still the same result. I think these boards are comparable to a LOLIN32.

elixx, 2019-05-14:
Trying with SPI firmware, I get a reboot and the following:
```

>>> ***ERROR*** A stack overflow in task mp_thread has been detected.
abort() was called at PC 0x40091d8c on core 1

Backtrace: 0x40091a3f:0x3ffe90e0 0x40091d75:0x3ffe9100 0x40091d8c:0x3ffe9120 0x40093e96:0x3ffe9140 0x40092d54:0x3ffe9170 0x40092d0a:0x3ffc49a0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4928
ho 0 tail 12 room 4
load:0x40078000,len:10188
load:0x40080400,len:6736
entry 0x400806e8
```

elixx, 2019-05-14:
I found that calling the same function from a route works totally find in threaded mode and not.
It appears to be something confined to the handling of pyHTML files (from what I can see so far).

jczic, 2019-05-16:
Hello @elixx and thanks to interesting:)
I know this problem with micropython because there are a large number of function penetrations and this can cause a stack overflow.
And with the pyhtml model, it's even more...
A solution is to access the micropython variables to increase the stack maximum value and recompile the firmware :o
Do you can try that for your needs ?

elixx, 2019-05-16:
Thanks -- I'll give that a shot.

pyHTML would be nice to get working, eventually on my board -- but I have a few different ones on the way, and see how it goes on those, too. Then I'll pursue rolling my own firmware. =)

Right now I switched to using the request handler decorators to get the functionality that I want -- so far, so good!

Thank you!

closed 2019-05-16 by elixx

jczic, 2019-10-20:
Hello,

I released a fully new version (v2.0) of my web server here :
github.com/jczic/MicroWebSrv2.
Open source MIT, fully asynchronous, more robust, more fast and more
efficient!
It is delivered with a good documentation.

Thank you for your support and feedback. ☺️

Jean-Christophe Bos