jczic/MicroWebSrv #55: HTTPS and WSS support

vtamas00, 2019-06-07:
Are you planning to extend the features with secure operation?
How big effort to implement the SSL handsake to this project?

dmeiburg, 2019-06-20:
I added SSL to  microWebSrv.py myself and experienced no issues so far.

Basically all that I did, was wrapping `client` from `client, cliAddr = self._server.accept()` with  `ussl.wrap_socket`. 

Here is my version of [microWebSrv.py](https://github.com/jczic/MicroWebSrv/files/3310855/microWebSrv.py.txt)
Note, that I also use a certificate and a key from separate files. 

[microWebSrv.py](https://github.com/jczic/MicroWebSrv/files/3310855/microWebSrv.py.txt)

Here is some output using curl to connect to an esp32 using SSL:

```
$ curl -v -k https://doorlock004/ -H "Authorization: Token ********************"                                                                                                                                                                                                       
* TCP_NODELAY set
* Connected to doorlock004 port 443 (#0)                                                                                                                                       
* ALPN, offering h2                                                                                                                                                                           
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*   CAfile: none
  CApath: /etc/ssl/certs
* TLSv1.3 (OUT), TLS handshake, Client hello (1):                                                                                                                                             
* TLSv1.3 (IN), TLS handshake, Server hello (2):                                                                                                                                              
* TLSv1.2 (IN), TLS handshake, Certificate (11):                                                                                                                                              
* TLSv1.2 (IN), TLS handshake, Server key exchange (12):                                                                                                                                      
* TLSv1.2 (IN), TLS handshake, Server finished (14):                                                                                                                                          
* TLSv1.2 (OUT), TLS handshake, Client key exchange (16):                                                                                                                                     
* TLSv1.2 (OUT), TLS change cipher, Client hello (1):
* TLSv1.2 (OUT), TLS handshake, Finished (20):                                                                                                                                                
* TLSv1.2 (IN), TLS handshake, Finished (20):                                                                                                                                                 
* SSL connection using TLSv1.2 / ECDHE-RSA-AES256-GCM-SHA384
* ALPN, server did not agree to a protocol
* Server certificate:
*  subject: C=DE; ST=MV; L=Rostock; O=Fraunhofer IGD-R; OU=ITSC; CN=doorlock004
*  start date: Feb  7 14:20:13 2019 GMT
*  expire date: Feb  7 14:20:13 2020 GMT
*  issuer: C=DE; ST=MV; L=Rostock; O=Fraunhofer IGD-R; OU=ITSC; CN=doorlock004
*  SSL certificate verify result: self signed certificate (18), continuing anyway.
> GET / HTTP/1.1
> Host: doorlock004
> User-Agent: curl/7.61.0
> Accept: */*
> Authorization: Token ********************
> 
< HTTP/1.1 200 OK                                                                                                                                                                             
< Content-Type: text/html; charset=UTF-8                                                                                                                                                      
< Content-Length: 230
< Server: MicroWebSrv by JC`zic                                                                                                                                                               
< Connection: close                                                                                                                                                                           
< 
        <!DOCTYPE html>
        <html lang=en>
        <head>
                <meta charset="UTF-8" />
            <title>I am doorlook004</title>
        </head>
        <body>
        <h1> I am doorlook004</h1>
        </body>
* Closing connection 0
* TLSv1.2 (OUT), TLS alert, Client hello (1):                                                                                                                                                 
        </html></body>
```

vtamas00, 2019-07-01:
Thanks, very nice, I will try it out!

jczic, 2019-10-20:
Hello,

I released a fully new version (v2.0) of my web server here :
github.com/jczic/MicroWebSrv2.
Open source MIT, fully asynchronous, more robust, more fast and more
efficient!
It is delivered with a good documentation.

Thank you for your support and feedback. ☺️

Jean-Christophe Bos

closed 2019-11-09 by jczic