jczic/MicroWebSrv #59: ESP32 - mws.IsStarted() false when threaded

xal88, 2019-07-26:
Hi,
when running  **mws.Start(threaded=True)**
then    **mws.IsStarted()**   always returns false. 

mws is working fine, but of course   **mws.Stop()**   also fails. 

What am I missing ? Thanks for your support
Alf
 *ESP32 4MB , micropython version='v1.11-178'*

jczic, 2019-07-26:
And if you wait a little after the Start call and before the IsStarted ?

xal88, 2019-07-27:
With a pause this is working fine.

```python
MicroPython v1.11-178-gad0b7cb01 on 2019-07-25; ESP32 module with ESP32
>>> from microWebSrv import MicroWebSrv
>>> mws = MicroWebSrv(webPath="/www")
>>> mws.Start(threaded=True)
>>> utime.sleep(10)
>>> mws.IsStarted()
True
```
So my 2 seconds wait was too short.  
How can I stop/restart the server now  ? - wait does not help there.
```python
>>> mws.Stop()
>>> utime.sleep(10)
>>> mws.IsStarted()
True
```

jczic, 2019-10-20:
Hello,

I released a fully new version (v2.0) of my web server here :
github.com/jczic/MicroWebSrv2.
Open source MIT, fully asynchronous, more robust, more fast and more
efficient!
It is delivered with a good documentation.

Thank you for your support and feedback. ☺️

Jean-Christophe Bos

closed 2019-11-09 by jczic