jczic/MicroWebSrv #61: 403 Forbidden error occured when loading the fonts folder of the webapp made with Vue.js

dukeduck1984, 2019-07-30:
Hi,

I'm using the MicroWebSrv to serve both the back end (rest api) and the front end (made with vue.js).  The back end worked fine, but there is some issues with the front end.

The web app packed by Vue.js has 4 folders along with index.html, they are: css, fonts, js, statics.  Altough extremely slow when loading the page for the first time, most parts of the page have loaded correctly except for the icons which are fonts.  So I took a look at the console, and found some errors, e.g. """GET http://192.168.123.108/fonts/flUhRq6tzZclQEJ-Vdg-IuiaDsNcIhQ8tQ.0509ab09.woff2 net::ERR_ABORTED 403 (Forbidden)"""

How can I solve this problem?  Is there a way to mark this folder as static folder like in Flask?

Thanks,
Kaiyuan

dukeduck1984, 2019-07-30:
Does it have something to do with the internally supported Mimetypes?  Is it possible to manually add other mimetypes?  For example, those font files are ended with .woff or .woff2

jczic, 2019-07-30:
I've solved the problem :)
You can pull the git (last commit https://github.com/jczic/MicroWebSrv/commit/425e665035fbee601f686c22107ade3dda0c76c0)

dukeduck1984, 2019-07-30:
Thank you so much!!

I was just about to say that I added 2 lines in _mimeTypes and now it's working.  Wonderful! :-))

closed 2019-07-30 by dukeduck1984

jczic, 2019-10-20:
Hello,

I released a fully new version (v2.0) of my web server here :
github.com/jczic/MicroWebSrv2.
Open source MIT, fully asynchronous, more robust, more fast and more
efficient!
It is delivered with a good documentation.

Thank you for your support and feedback. ☺️

Jean-Christophe Bos