jczic/MicroWebSrv #62: Error 112 on ESP32

arpadtamasi, 2019-07-31:
During development every time I download a new version, I got the following trace:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<string>", line 34, in <module>
  File "microWebSrv.py", line 224, in Start
OSError: 112

amotl, 2019-07-31:
Hi there,

first things first: Thanks Jean-Christophe for conceiving this excellent piece of software!

While developing the Terkin Datalogger [1] (as well on ESP32 with Pycom MicroPython), we probably experienced the same thing as @arpadtamasi:
```python
   22.3080 [terkin.api.http          ] INFO   : Setting up HTTP API
   22.4902 [terkin.api.http          ] INFO   : Starting HTTP server
Traceback (most recent call last):
  File "main.py", line 65, in <module>
  File "main.py", line 60, in main
  File "/flash/lib/terkin/datalogger.py", line 130, in start
  File "/flash/lib/terkin/device.py", line 211, in start_network_services
  File "/flash/lib/terkin/network/core.py", line 70, in start_httpserver
  File "/flash/lib/terkin/api/http.py", line 54, in start
  File "dist-packages/microWebSrv.py", line 224, in Start
OSError: [Errno 12] ENOMEM
```
-- https://github.com/hiveeyes/hiveeyes-micropython-firmware/blob/0deb953a4d241bb78ccf6aae76c932bcbb9fe95b/doc/troubleshooting.rst#sometimes-the-web-server-runs-out-of-memory-crashing-the-system

We compensated for that by checking for `webserver.IsStarted()` appropriately when starting the server in a new thread like `webserver.Start(threaded=True)`.
```python
def start(self):
    if self.webserver.IsStarted():
        log.info('HTTP server already started')
    else:
        log.info('Starting HTTP server')
        self.webserver.Start(threaded=True)
```
-- https://github.com/hiveeyes/hiveeyes-micropython-firmware/blob/0deb953a4d241bb78ccf6aae76c932bcbb9fe95b/terkin/api/http.py#L65-L70

So, the reason for your observation could be that the code might be attempting to start the webserver twice.

With kind regards,
Andreas.

[1] https://github.com/hiveeyes/hiveeyes-micropython-firmware

arpadtamasi, 2019-08-02:
@amotl thanks, you are right. It happens to me when - during development - I download a new version of my main.py to ESP32. In this case I don't have access to the variable declared in the old version so cannot call IsStarted

amotl, 2019-08-02:
Dear Árpád,

thanks for confirming these observations and also thanks for clarifying what you mean by "download" - I already became curious to ask ;].

In order to work around the "lost reference" problem, we just declared a global variable for holding that reference. For us, it is a "module-wide global" variable, probably living inside the namespace of `terkin.api.http`.

This module-wide global variable just called `webserver` [1] is assigned like outlined within [2].

Maybe this helps.

With kind regards,
Andreas.

[1] https://github.com/hiveeyes/hiveeyes-micropython-firmware/blob/0deb953a4d241bb78ccf6aae76c932bcbb9fe95b/terkin/api/http.py#L30-L31
[2] https://github.com/hiveeyes/hiveeyes-micropython-firmware/blob/0deb953a4d241bb78ccf6aae76c932bcbb9fe95b/terkin/api/http.py#L57-L63

arpadtamasi, 2019-08-02:
Thanks, I give it a shot. I have no experience with Python neither with ESP that's why I'm confused. I'm just building a toy robot with my kids.

amotl, 2019-08-02:
> Thanks, I give it a shot. I have no experience with Python neither with ESP that's why I'm confused.

I see. Just throwing in a webserver sounds easy, right? However, I believe you will make it. Just forget about the module scope, stay with your `main.py` and declare your variable as `global` right where you are assigning it to be able to reference it even after a soft reboot usually being used while developing on the firmware.

> I'm just building a toy robot with my kids.

I see. Good luck!

amotl, 2019-08-02:
> when you are hitting a wall there, not seeing any exceptions raised from the handler methods.

```diff
$ diff -u dist-packages/microWebSrv.py var/microWebSrv.py | colordiff
--- dist-packages/microWebSrv.py	2019-08-02 17:11:03.000000000 +0200
+++ var/microWebSrv.py	2019-08-02 17:14:50.000000000 +0200
@@ -375,6 +375,7 @@
                     else :
                         response.WriteResponseBadRequest()
             except :
+                print('ERROR handling webserver request\n', get_last_stacktrace())
                 response.WriteResponseInternalServerError()
             try :
                 if self._socketfile is not self._socket:
@@ -861,3 +862,12 @@
     # ============================================================================
     # ============================================================================

+
+import sys
+import uio
+
+def get_last_stacktrace():
+    buf = uio.StringIO()
+    exc = sys.exc_info()[1]
+    sys.print_exception(exc, buf)
+    return buf.getvalue()
```

joseshiru, 2019-10-16:
Hi amotl, 

I tried to follow your work around but it didn't work.

My code looks like this: 
`global webserver
webserver = MicroWebSrv()
webserver.Start(threaded=True) # Starts server in a new thread`

Getting: 

ile "<stdin>", line 27, in <module>
  File "util/microWebSrv.py", line 230, in Start
OSError: 112

jczic, 2019-10-17:
Same TCP port already open ?
https://forum.micropython.org/viewtopic.php?t=4029

joseshiru, 2019-10-17:
I've tried with different ports without any success.

There's nothing else in my code using the same port.

Thanks.

jczic, 2019-10-17:
Ok, it's strange that you have this error 112.
Just: a very new version of my micro web server will be released for MicroPython in a few days!
For the moment, it works on Python (normal) and you can test it.
At the moment, repository doesn't have description, keywords etc... but a great documentation is present :)
I must fully test it on microPython, however,  it must be works on it.
Good luck, the link : https://github.com/jczic/MicroWebSrv2
👍

jczic, 2019-10-20:
Hello,

I released a fully new version (v2.0) of my web server here :
github.com/jczic/MicroWebSrv2.
Open source MIT, fully asynchronous, more robust, more fast and more
efficient!
It is delivered with a good documentation.

Thank you for your support and feedback. ☺️

Jean-Christophe Bos

joseshiru, 2019-10-20:
Perfect mate,

Gonna give it a try.

Thanks a lot.

Jose Zuma

*Sincèrement,*