jczic/MicroWebSrv #66: No logging is implemented to allow INFO,DEBUG,WARNING,ERROR and so.

nherriot, 2019-09-18:
More of a request of a feature than an error.... :-) 
It would be great if there was a logging utility to allow logging to be added easily to the web server.
Something like the normal logging utility for python e.g.

>>> import logging
>>> logging.basicConfig(level=logging.INFO)
>>> log = logging.getLogger("microWebSrv")
>>> log.debug("Failed message: %d(%s)", 100, "HTTP 500")

Kind regards, Nicholas.

jczic, 2019-09-20:
Hi,
Yes, good idea but microWebSrv must works on embedded platforms without that ... (with micropython).

closed 2019-09-27 by jczic

jczic, 2019-10-20:
Hello,

I released a fully new version (v2.0) of my web server here :
github.com/jczic/MicroWebSrv2.
Open source MIT, fully asynchronous, more robust, more fast and more
efficient!
It is delivered with a good documentation.

Thank you for your support and feedback. ☺️

Jean-Christophe Bos

nherriot, 2019-10-21:
I'll have a look.
But I'm really happy with V1 :-) ....
I've added logging etc... and promise to do a pull request for you to
evaluate.