jczic/MicroWebSrv #70: SendText

gpspelle, 2020-04-01:
Hi, 

After reading this incredible Library, I am not sure yet about the send message process. As far as I understood, the example that you gave in your main.py file, you send a reply whenever you have received a message. But how should I do to send a message even I didn't get a message in the first place?

Thanks!

jczic, 2020-04-01:
Hello @gpspelle,
You can check the `OnWSChatTextMsg()` function in `main.py` who is used to send msg to all WebSockets connections (for the chat example) registered in `_chatWebSockets` list.
You can first register websockets object (connections) in an array and after, send messages when you want (by a timer or any event).
It's ok for you?

gpspelle, 2020-04-01:
Yeah, I'll try to use it, but I don't know if this will solve my problem in the end.

jczic, 2020-04-02:
And what's your problem exactly?

Kiri-fr, 2020-06-12:
Hi Gabriel, I had the same issue.
What you could do is: 

def _acceptWebSocketCallback(webSocket, httpClient) :
    print("WS ACCEPT")
    global WS
    WS = webSocket   # Save socket access point for external use
    webSocket.RecvTextCallback = _recvTextCallback
    webSocket.RecvBinaryCallback = _recvBinaryCallback
    webSocket.ClosedCallback = _closedCallback

WS.SendText("Hi")

This code was proposed by Marc Fiot

closed 2021-07-04 by gpspelle