jczic/MicroWebSrv #71: Websockets

gpspelle, 2020-04-01:
Hi, 

I have this Django project running some websockets for internal communication. I'd like to know more about how to connect to them with your implementation of websockets. In normal python I'd do:

`connect("ws://gpspelle.com/ws/chat/ble")`

In your code, how'd it be? 

`MicroWebSrv(bindIP='gpspelle.com', webPath='/ws/chat/ble')`

I am sorry if those are too basic questions. Thanks anyway.

gpspelle, 2020-04-01:
Also, I've just realized that in 2018 there were no client websocket yet.
https://github.com/jczic/MicroWebSrv/issues/25

jczic, 2020-04-02:
Please, use MicroWebSrv2 : https://github.com/jczic/MicroWebSrv2.
There is help, and examples for do that :)

closed 2020-04-02 by jczic