jczic/MicroWebSrv #78: WebSocket Critical Question ?

thevobos, 2023-03-31:
Hi, Thank You for created the this package.

How can i connect to my socket.io server my server ?.

jczic, 2023-03-31:
Hello and thank you @thevobos.
MicroWebSrv is an HTTP(s) server with a plugin for WebSockets but it's not a "client".
So, if you want to connect 2 serveurs, you must use a client "in" the server :)
That's ok for you?

thevobos, 2023-03-31:
Thank you for your quick response.

socket.io could not find library to connect. Any chance of guidance on this?

jczic, 2023-03-31:
There are not many websockets "clients" for micropython unlike python.
The most used one seems to be Danni's but indeed it is proposed for esp8266.
You can read this : **https://github.com/danni/uwebsockets/issues/7**